bitbank2/bb_spi_lcd #4: How did you create the gif.h?

jhonnyx161, 2021-04-11:
Hello
How did you create the gif.h?
regards

jhonnyx161, 2021-04-11:
> Hello
> How did you create the gif.h?
> regards

closed 2021-04-11 by jhonnyx161

bitbank2, 2021-04-11:
I used my image_to_c tool:

https://github.com/bitbank2/image_to_c

jhonnyx161, 2021-04-12:
> I used my image_to_c tool:
> 
> https://github.com/bitbank2/image_to_c
Thank you so much

how to connect
st7789 240*240 without cs pin and mini sd bb_spi_lcd

bitbank2, 2021-04-12:
This question is beyond the scope of AnimatedGIF. I believe I have an example with that display in my bb_spi_lcd examples.

jhonnyx161, 2021-04-12:
> Esta pregunta está más allá del alcance de AnimatedGIF. Creo que tengo un ejemplo con esa pantalla en mis ejemplos bb_spi_lcd.

the example is with adafruit. Adafruit is slower than bb_spi_lcd?

I see in the debb_spi_lcd.cpp code this if:
if (pLCD-> iLCDType == LCD_ST7789 || pLCD-> iLCDType == LCD_ST7789_240 || pLCD-> iLCDType == LCD_ST7789_135 || pLCD-> iLCDType == LCD_ST7789_NOCS)

I tried each one in spilcdInit () but with no positive results
am I missing something?
thanks for the support

jhonnyx161, 2021-04-12:
> Esta pregunta está más allá del alcance de AnimatedGIF. Creo que tengo un ejemplo con esa pantalla en mis ejemplos bb_spi_lcd.

the example uses st7789 with cs pin, it is another model

the example with adafruit and st7735 is slower than bb_spi_lcd?

notice debb_spi_lcd.cpp code this if:
if (pLCD-> iLCDType == LCD_ST7789 || pLCD-> iLCDType == LCD_ST7789_240 || pLCD-> iLCDType == LCD_ST7789_135 || pLCD-> iLCDType == LCD_ST7789_NOCS)

I tried with each mode in spilcdInit () but got no positive results for st7789 if cs pin

thank you ppr the support

bitbank2, 2021-04-12:
The correct choice is LCD_ST7789_NOCS. This is for the 240x240 display without an exposed CS pin. It needs the signal to be a phase of SPI mode 3.