bitbank2/bb_spi_lcd #5: rlottie widget for bb_spi_lcd

ajsb85, 2021-06-02:
Allows to use rlottie animations in the BitBank SPI Color LCD/OLED library

You can use [Samsung/rlottie](https://github.com/Samsung/rlottie) library with C-api header

Best regards.

Ref:
https://github.com/Samsung/rlottie/issues/484

bitbank2, 2021-06-02:
I don't see what rlottie has to do with my SPI LCD library. This code is for sending pixels to SPI displays with a few primitive functions. You're welcome to write application code to combine this with other projects, but it doesn't make any sense to integrate rlottie into bb_spi_lcd.

closed 2021-06-02 by bitbank2

ajsb85, 2021-06-02:
Hi, Larry .
Nice to read from you.
I will build a project and share with you here.
Cheers,