bitbank2/bb_spi_lcd #14: Errors trying to compile and run on ESP32

rsiemens77, 2024-01-30:
I have my ST7789 LCD setup and working with my ESP32 board with the TFT_eSPI library.  But, I want to support multiple displays, so without changing anything in my working hardware set up, I'm trying this library.  But I'm having trouble getting it to work.

The first issue I encountered is that the parallel file does not compile.  I get numerous errors, some of the most obvious are unknown types, such as:

lcd_cam_dev_t
esp_lcd_i80_bus_handle_t

I noticed there's references to S3 in this file, is this perhaps only for S3?  I removed references to the parallel API in the main file, and then I was able to compile.  I'm only using SPI anyway.

But now I get an exception when trying to init, from this call:

spilcdInit(&lcd, LCD_ST7789, FLAGS_NONE, 40000000, -1, 21, 19, -1, -1, 22, 23);

0x400dd0a3: spilcdWriteCommand(tagSPILCD*, unsigned char) at C:/exp/frameworks/esp-idf-v4.4.6/components/arduino/libraries/bb_spi_lcd/src/bb_spi_lcd.cpp:2909
0x400dd336: spilcdInit(tagSPILCD*, int, int, int, int, int, int, int, int, int, int) at C:/exp/frameworks/esp-idf-v4.4.6/components/arduino/libraries/bb_spi_lcd/src/bb_spi_lcd.cpp:1951
...

bitbank2, 2024-01-30:
Thanks for letting me know about the issues. The error you're getting is probably because of the changes I made for parallel. You should use the C++ API, not the C API. Try setting the lcd structure to 0's before calling spilcdInit(). I'll work on these 2 issues.

rsiemens77, 2024-01-30:
I was following the spi_lcd_demo example.  Are there any C++ examples?  Anyway, I switched to BB_SPI_LCD class and did get that working.  I can fill and draw to the LCD, but haven't been able to successfully display an image yet.

I tried pushImage with my lcd-image-converter as described in the below header.  That results in exception, seems to be this:

```
        ret=spi_device_polling_transmit(spi, &t);  //Transmit!
        assert(ret==ESP_OK);            //Should have had no issues.

assert failed: void myspiWrite(SPILCD*, unsigned char*, int, int, int) bb_spi_lcd.cpp:1579 (ret==ESP_OK)

0x400dd9ce: myspiWrite(tagSPILCD*, unsigned char*, int, int, int) at C:/exp/frameworks/esp-idf-v4.4.6/components/arduino/libraries/bb_spi_lcd/src/bb_spi_lcd.cpp:1579 (discriminator 1)
0x400dda5d: spilcdWriteDataBlock(tagSPILCD*, unsigned char*, int, int) at C:/exp/frameworks/esp-idf-v4.4.6/components/arduino/libraries/bb_spi_lcd/src/bb_spi_lcd.cpp:1646
0x400dfb4b: BB_SPI_LCD::pushImage(int, int, int, int, unsigned short*, int) at C:/exp/frameworks/esp-idf-v4.4.6/components/arduino/libraries/bb_spi_lcd/src/bb_spi_lcd.cpp:6011
```

```
/*******************************************************************************
* generated by lcd-image-converter rev.030b30d from 2019-03-17 01:38:34 +0500
* image
* filename: unsaved
* name: alien154x190-straight
*
* preset name: Color R5G6B5
* data block size: 16 bit(s), uint16_t
* RLE compression enabled: no
* conversion type: Color, not_used not_used
* split to rows: yes
* bits per pixel: 16
*
* preprocess:
*  main scan direction: top_to_bottom
*  line scan direction: forward
*  inverse: no
*******************************************************************************/
static const uint16_t alien_straight_bm[29260] PROGMEM = {
```

bitbank2, 2024-01-31:
There are 3 c++ examples:
UNO_shield_demo
atoms3_gif
parallel_shield_demo
You can also look at the Wiki of this repo for details about all of the class methods.

Did clearing the C structure to 0 before calling spilcdInit() work?

rsiemens77, 2024-02-01:
I'm using the C++ api now as you suggested, but I went back and tried C for you.  Yes, I could start if I initialize to 0.

Where is the wiki?  I can't find it.

bitbank2, 2024-02-01:
ooh - you're right about the lack of Wiki. I haven't given this repo as much attention because it seemed like few people were using it. I'll correct that shortly.

rsiemens77, 2024-02-29:
I gather it's few people are using it, because you haven't paid much attention to it.  It doesn't compile on the most common ESP32 platform, and when hacked to make it compile it has runtime bugs.

bitbank2, 2024-02-29:
Open Source projects are unfortunately on hobby time because I have a family to feed. When someone pays me to make this higher priority it will get faster fixes. It works for my projects on my copy of the repo. I would really like for people to use this code instead of TFT_eSPI which is excessively awful for newbies because of its dumb .H file config. The problem is that I'm not independently wealthy, so priorities shift to where I will get paid.

bitbank2, 2024-02-29:
I pushed a fix, please give it a try.

guedesite, 2024-03-22:
Hi @rsiemens77, I'm in the same situation as you are. I would like to use these libraries. Do they work properly now? Have you noticed a significant performance gain?

guedesite, 2024-03-22:
I was disappointed, but I still gave it a shot, and surprise... the compilation went smoothly, the demo runs without any issues on an unofficial esp32-wroom-UE 16N, well done! Despite the lack of a wiki, I find it much simpler and easier to use with a basic understanding of coding compared to TFT_eSPI, which I think is too heavy for certain applications. It's exactly what I needed.