bitbank2/bb_spi_lcd #15: Crash when trying to draw image on ESP32

rsiemens77, 2024-01-30:
I am using BB_SPI_LCD class on ESP32 with ST7789.  I can fill and draw to the LCD, but haven't been able to successfully display an image.

I tried lcd.pushImage(0,0,154,190,alien_straight_bm); with my lcd-image-converter output as described in the below header. That results in this exception:

        ret=spi_device_polling_transmit(spi, &t);  //Transmit!
        assert(ret==ESP_OK);            //Should have had no issues.

assert failed: void myspiWrite(SPILCD*, unsigned char*, int, int, int) bb_spi_lcd.cpp:1579 (ret==ESP_OK)

0x400dd9ce: myspiWrite(tagSPILCD*, unsigned char*, int, int, int) at bb_spi_lcd/src/bb_spi_lcd.cpp:1579 (discriminator 1)
0x400dda5d: spilcdWriteDataBlock(tagSPILCD*, unsigned char*, int, int) at bb_spi_lcd/src/bb_spi_lcd.cpp:1646
0x400dfb4b: BB_SPI_LCD::pushImage(int, int, int, int, unsigned short*, int) at libraries/bb_spi_lcd/src/bb_spi_lcd.cpp:6011

/*******************************************************************************
* generated by lcd-image-converter rev.030b30d from 2019-03-17 01:38:34 +0500
* image
* filename: unsaved
* name: alien154x190-straight
*
* preset name: Color R5G6B5
* data block size: 16 bit(s), uint16_t
* RLE compression enabled: no
* conversion type: Color, not_used not_used
* split to rows: yes
* bits per pixel: 16
*
* preprocess:
*  main scan direction: top_to_bottom
*  line scan direction: forward
*  inverse: no
*******************************************************************************/
static const uint16_t alien_straight_bm[29260] PROGMEM = {

bitbank2, 2024-01-31:
It looks like it may have something to do with the SPI driver not liking to write data from FLASH memory. Can you try removing the 'const' and 'PROGMEM' labels and see if it works? Everything else looks ok.

rsiemens77, 2024-02-01:
I removed those, it did not make a difference.  I did notice some additional output (happens either way) on the log:

`E (1559) spi_master: check_trans_valid(696): txdata transfer > host maximum`

rsiemens77, 2024-02-01:
Removing those does not make a difference.  I did notice some log error I didn't report before (happens either way):

`E (1559) spi_master: check_trans_valid(696): txdata transfer > host maximum`

bitbank2, 2024-02-01:
Ah - that's the issue. Can you send me a complete example which reproduces the problem? I'll make sure it works and push a fix. Thanks.

rsiemens77, 2024-02-01:
The code is in the first message, it's just one call.  If you need an image to test with, I attached it.

`lcd.pushImage(0,0,154,190,alien_straight_bm);`

[alien_straight.zip](https://github.com/bitbank2/bb_spi_lcd/files/14130816/alien_straight.zip)

rsiemens77, 2024-02-11:
Any word on fixing this?

rsiemens77, 2024-02-29:
It's unfortunate this is buggy and no longer being maintained.

bitbank2, 2024-02-29:
It's unfortunate that people can only complain about free software and not contribute.

bitbank2, 2024-02-29:
I pushed a fix for this, give it a try.

rsiemens77, 2024-03-22:
The problem was you initially responded in a timely fashion like you'd look into it, then no word for a month.  I did my part, testing and providinig feedback.  When I then heard nothing for a month, I had to move on, thus my furstrated post.