bitbank2/bb_spi_lcd #17: DRAW_WITH_DMA only draw half of the screen

guedesite, 2024-03-23:
Hi, I am working on an ESP32-WROOM-32UE with arduino-esp32 and an LCD_ST7789_240 (40MHz) screen. 

This bug occurs with the spilcdWriteDataBlock function and the DRAW_WITH_DMA flag, tested with the JPEGDEC example [esp32_jpeg](https://github.com/bitbank2/JPEGDEC/blob/master/examples/esp32_jpeg/esp32_jpeg.ino). 

My image, adapted to the resolution of my screen, has the following description: 
```
// Data size = 4213 bytes
//
// JFIF, Compression=JPEG, Size: 240 x 240, 24-Bpp
//
```

Changing the resolution, the quality of my JPEG, or the SPI frequency didn't make a difference. 

I don't have the capability to manually test if the problem is related to JPEG decoding or not. However, I did test the PNGdec png_transparency example, and spilcdWritePixelsMasked functions correctly with DRAW_WITH_DMA.

[image]

bitbank2, 2024-03-23:
Can you share the complete sketch that you're running, along with the exact image? From the description and photo, I don't have enough info to debug the problem.

guedesite, 2024-03-23:
@bitbank2 here is it 
[test_jpeg.zip](https://github.com/bitbank2/bb_spi_lcd/files/14731275/test_jpeg.zip)

bitbank2, 2024-03-23:
I was able to reproduce the problem and am working on a fix. It has to do with the max transaction size of a DMA write (4092 bytes).

bitbank2, 2024-03-23:
ok, I pushed new code to fix it. Be advised that I changed the spilcdInit function to add a new parameter - bUseDMA. I needed this for the SD card to work on the M5Stack Core2. I recommend using the C++ api instead of the old C API.

closed 2025-04-29 by guedesite