bitbank2/bb_spi_lcd #23: 2.6.0 does not compile for #defined ARDUINO_ESP32S3_DEV

cyberman54, 2024-08-04:
[{
	"resource": "bb_spi_lcd/src/bb_parallel.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'struct esp_lcd_rgb_panel_config_t' has no member named 'num_fbs'",
	"startLineNumber": 370,
	"startColumn": 17,
	"endLineNumber": 370,
	"endColumn": 17
},{
	"resource": "bb_spi_lcd/src/bb_parallel.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'struct esp_lcd_rgb_panel_config_t' has no member named 'bits_per_pixel'",
	"startLineNumber": 374,
	"startColumn": 17,
	"endLineNumber": 374,
	"endColumn": 17
},{
	"resource": "bb_spi_lcd/src/bb_parallel.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'esp_lcd_rgb_panel_get_frame_buffer' was not declared in this scope",
	"startLineNumber": 413,
	"startColumn": 4,
	"endLineNumber": 413,
	"endColumn": 4
}]

bitbank2, 2024-08-04:
The latest bb_spi_lcd has functions which require the latest (3.0.x) Arduino ESP32 support (IDF V5.x). Are you using the latest?

cyberman54, 2024-08-04:
No. My App is on Arduino 2.0.4, and i can't migrate to 3.0.x.due to IRAM shortage.

What is the latest release of this lib working with pre Arduino 3.0.x? I will nail my app on that release, then.

cyberman54, 2024-08-04:
I checked my current versions:
- Arduino v2.0.11 ([release notes](https://github.com/espressif/arduino-esp32/releases/tag/2.0.11))
- ESP-IDF v5.1.1 ([release notes](https://github.com/espressif/esp-idf/releases/tag/v5.1.1))
With these frameworks in run into the compiler error with bb_spi_lcd lib.

bitbank2, 2024-08-04:
Probably release 2.4.1 is the one that will work for you.

closed 2024-08-04 by bitbank2