bitbank2/bb_spi_lcd #24: DMA option for BB_SPI_LCD::begin to disable DMA when using SD card module on the same SPI bus

thelastoutpostworkshop, 2024-08-07:
Currently the function BB_SPI_LCD::begin is using DMA by default.
I am using an SD card on the same SPI bus as my display.

bitbank2, 2024-08-09:
It's probably not the DMA that's the problem,  but re-initializing the SPI bus in half-duplex mode. I don't have a solution for you at the moment. My code supports sharing the SPI with a touch controller, but not the display part. If you're brave, you can modify the code for your scenario and then do a pull request.

closed 2024-08-09 by bitbank2

thelastoutpostworkshop, 2024-08-09:
I switched to using SPILCD class, I did not need to re-initialize the SPI bus, it works like a charm!

thelastoutpostworkshop, 2024-08-09:
And I disabled DMA by passing 0 :     spilcdInit(&tft, LCD_ILI9341, FLAGS_NONE, TFT_FREQUENCY, TFT_CS_PIN, TFT_DC_PIN, TFT_RST_PIN, -1, -1, TFT_MOSI_PIN, TFT_CLK_PIN, 0);

bitbank2, 2024-08-09:
Yes, if you don't mind the C style interface, then you can use that. I do plan on deprecating it in the near future.

thelastoutpostworkshop, 2024-08-09:
If I enable DMA in spilcdInit, the SD module is not working anymore.
So could it be as simple as adding the option to BB_SPI_LCD::begin ?

bitbank2, 2024-08-09:
Yes, the underlying code is the same