bitbank2/bb_spi_lcd #25: How to draw on the backbuffer

ps1x, 2024-09-28:
Hello and thank you for your great work!

I have not found example or any info (and honestly i'm too stupid to understand) on how to first draw some pictures (BMPs) and text and primitives on some buffer and then output buffer contents to LCD?

Second question is about lcd.drawBmp, somehow my BMPs looks like "overscan" images, sort of choppy lines like in 90s VHS recordings, i use that round 360x360  aluminium aliexpress display with SD card and audio output.

Thank you!

bitbank2, 2024-09-29:
This isn't an issue, but I'll answer your question. When there is no back buffer, the default drawing flags are "DRAW_TO_LCD". This sends all pixels directly to the LCD controller. When you allocate a back buffer, you can send the drawing commands to the backbuffer, the LCD or both. The drawing flags are DRAW_TO_RAM, and DRAW_TO_LCD. They can be used individually or together. To send the backbuffer contents to the LCD, use the C++ method display() or the C method spilcdShowBuffer() - they're the same.
The drawBMP() method takes a Windows BMP file as input. If you've got a supported file that doesn't display correctly, please send me a copy of the image.

ps1x, 2024-09-30:
[digits.zip](https://github.com/user-attachments/files/17186069/digits.zip)
I attached my BMP file together with h file created with xxd linux command. Thank you!

ps1x, 2024-09-30:
New info: i used buffer, and its all fine with image quality now. Very strange.

closed 2024-09-30 by bitbank2

ps1x, 2024-09-30:
I'm sorry to bother you but i'm struggling with another question, and maybe other users will search this too: i want to keep FPS high, so i don't want to redraw whole background every frame, in order to do this i want to store some part of the screen i'm about to change into buffer, then draw feature. Next frame instead of redrawing whole background i want to restore just that buffer underneath the feature. Then cycle continues. I see there is lcd.captureArea method, but i get weird results using it. Is it correct to use it like this:
`
u16_t pixels[90]; // buffer for pixels
  lcd.captureArea(0,0, x, y, 30, 30, pixels); // save 30 by 30 squre of pixels from backbuffer to pixels variable
lcd. fillCircle(x,y, 15,TFT_RED); // draw circle at the same location
lcd.pushPixels(pixels, 90); // restore image 30x30=90 pixels as it was before drawing circle
`
Thank you!

bitbank2, 2024-10-01:
You can create a virtual display (just RAM) of any size, draw in it using the normal functions and then draw that buffer onto an LCD. For C++ you will use 2 instances of the BB_SPI_LCD class.