bitbank2/bb_spi_lcd #26: Alternative licence to GPL

PhonicUK, 2024-11-11:
Being straight GPL makes this library unusable since it then cannot be used in anything other than other GPL licenced projects. Please consider licencing under something more friendly such as LGPL or MIT.

bitbank2, 2024-11-12:
Can you be more specific as to why this is an issue for you? Do you have a professional use case that the license is preventing you from using it?

PhonicUK, 2024-11-13:
I'm building an ESP32 based retro-style game system which I hope to be able to sell in the future (And more to the point - sell or provide a store for games on it). Right now I'm using TFT_eSPI (MIT licenced) but it has some issues with DMA which is why I'm considering this library. But as it currently stands I can't even consider this library because I can't use it unless I'm prepared to give away 100% of the source code for every game, application and utility running on it that uses this library. That might be fine for small example projects or tutorials but GPL makes it untouchable for serious work. It's virtually unheard of for libraries to be GPL because it's a viral licence. The libraries I've written and made available to the public similarly are MIT licenced so they are unencumbered.

bitbank2, 2024-11-13:
Thanks for clarifying your position on it. I've justifiably been a bit cynical of open source use since commercial enterprises have used my code without attribution and profited without even supporting my work with a token donation. I don't have a problem switching the license type and I hope you'll consider becoming a sponsor of my work if it gets into a commercial version of your project.

closed 2024-11-13 by bitbank2

PhonicUK, 2024-11-13:
I certainly understand your concern. It's a bit of a double-edged sword in some ways. On the one hand you do indeed run the risk that someone uses it in a very successful project and doesn't reward your effort, on the other you limit your user base to people who will give their software away for free and are in less of a position to sponsor. [My company](https://github.com/CubeCoders/) absolutely sponsors projects and libraries that we use in production :)

I appreciate your consideration and switching to Apache 2.0, that's very decent of you!