bitbank2/bb_spi_lcd #27: Any plans to enable setTextSize with build in fonts?

EricInMi, 2024-11-22:
I saw your article in Make: volume 91 and have found the library very useful.  I'd like to be able to scale up the built in font's.  Are there any plan to enable setTextSize?

bitbank2, 2024-11-23:
Thanks for letting me know where you find the code; I'm glad you find it useful. I don't like the "Adafruit" way of drawing text 1 pixel at a time, and then having huge, blocky characters. I support custom fonts (Adafruit_GFX format) that are for larger text sizes.

closed 2024-11-23 by bitbank2

EricInMi, 2024-11-26:
I'm now trying to use one two of the free fonts from the Adafruit_GFX library.

#include "FreeSans12pt7b.h"
#include "FreeSans24pt7b.h"

I'm getting an error when I compile.
C:\Users\Eric\Documents\Arduino\GPS_Lap_Timer_CYB\GPS_Lap_Timer_CYB.ino: In function 'void setup()':
C:\Users\Eric\Documents\Arduino\GPS_Lap_Timer_CYB\GPS_Lap_Timer_CYB.ino:86:15: error: cannot convert 'const GFXfont' to 'int'
   86 |   lcd.setFont(FreeSans12pt7b);
      |               ^~~~~~~~~~~~~~
      |               |
      |               const GFXfont
In file included from C:\Users\Eric\Documents\Arduino\GPS_Lap_Timer_CYB\GPS_Lap_Timer_CYB.ino:31:
c:\Users\Eric\Documents\Arduino\libraries\bb_spi_lcd\src/bb_spi_lcd.h:214:22: note:   initializing argument 1 of 'void BB_SPI_LCD::setFont(int)'
  214 |     void setFont(int iFont);

Any suggestions?

..and thank you.

bitbank2, 2024-11-26:
The function you're looking for is setFreeFont()