bitbank2/bb_spi_lcd #28: Large fonts don't show on cyd 22

u13726, 2024-12-23:
When using a large font on a cyd 22, only a small part is rendered. The same code works well on a cyd 2USB.
```

 #define CYD_28R_2USB
#include <bb_spi_lcd.h>
// #define CYD_22C
//#define CYD_28R_2USB

// 2.2" 240x320 ST7789 parallel LCD w/cap touch
#ifdef CYD_22C
//#define TOUCH_CAPACITIVE
#define TOUCH_SDA 21
#define TOUCH_SCL 22
#define TOUCH_INT -1
#define TOUCH_RST -1
#define LCD DISPLAY_CYD_22C
#endif
#ifdef CYD_28R_2USB
#define TOUCH_RESISTIVE
#define LCD DISPLAY_CYD_2USB
#endif

#include <bb_spi_lcd.h>
#include "Roboto_Thin66pt7b.h"
 BB_SPI_LCD lcd;
#define FONT Roboto_Thin66pt7b
#define FONT_GLYPHS Roboto_Thin66pt7bGlyphs

void setup() {
lcd.begin(LCD);
lcd.fillScreen(TFT_BLACK);
lcd.setFreeFont(&FONT );
lcd.setTextColor(TFT_GREEN, TFT_BLACK);
lcd.setCursor(50,200);
lcd.print("12");
 } /* setup() */
 
void loop() {
 delay(30000);
} /* loop() */
```

[image]

bitbank2, 2024-12-24:
Check the font data. Adafruit's font converter uses 8-bit signed integers for the x/y character offsets and overflows when the character size is too large. Look for negative numbers in the font info to see if this is the issue.

u13726, 2024-12-25:
Didn't find any negative values so far. Tested a bit more. And the issue occurs when displaying a character with over 128 bytes of bitmap info.

bitbank2, 2024-12-25:
The bitmap size shouldn't be an issue. Can you send me a font file that fails?

u13726, 2024-12-25:
This font has the problem. Characters '<>' are below the limit and ok ,'ST'
are just above .

bitbank2, 2024-12-25:
Sorry, your last message didn't work. Can you email the font font? Send to bitbank@pobox.com

Thanks

u13726, 2024-12-25:
Changing the limit governing the fill level of the buffer ucRXbuf in
bb_spi_lcd.h (line 4654 changed 4 to 16) bypasses the problem, maybe at
some speed cost. When the buffer was filled till 2400 bytes error occurred,
now filled max at 512 full font displays also on   CYD22
           for (ty=0; ty<cy; ty++) {
            for (tx=0; tx<pGlyph->width; tx++) {
               if (bits == 0) { // need to read more font data
                  uc = pgm_read_byte(&s[iBitOff>>3]); // get more font
bitmap data
                  bits = 8 - (iBitOff & 7); // we might not be on a byte
boundary
                  iBitOff += bits; // because of a clipped line
                  uc <<= (8-bits);
                  k = (int)(d-(uint16_t *)ucRXBuf); // number of words in
output buffer
                  if (k >= sizeof(ucRXBuf)/*16/*4*/*) { // time to write it
                     myspiWrite(pLCD, ucRXBuf, k*sizeof(uint16_t),
MODE_DATA, iFlags);
                     d = (uint16_t *)ucRXBuf;
                  }
               } // if we ran out of bits
               if (tx < cx) {
                  *d++ = (uc & 0x80) ? usFGColor : usBGColor;
               }
               bits--; // next bit
               uc <<= 1;
            } // for tx
            } // for ty

bitbank2, 2024-12-26:
The sample code you sent me uses the API incorrectly. The library is drawing the characters properly.

closed 2024-12-26 by bitbank2