bitbank2/bb_spi_lcd #36: tft.pushImage reduced  amout of bytes for transfer

Chris01011964, 2025-04-01:
Hello 

every lib version over 2.8.1 in combination with PSRAM just allow a maximum of approx 5000 byte for direct transfer. With the lib 2.8.1 was more than 3 times possible. I use the Guition JC482W543C with the NV3041A 

exeption occurs if you use a sketch which was working with 2.8.1 und with the newest Version of the lib. (2.9.3)

E (1182) spi_master: check_trans_valid(1053): txdata transfer > host maximum

regards
Chris

bitbank2, 2025-04-03:
Can you elaborate a bit on the issue? What do you mean by "direct transfer"? Do you have a simple example to reproduce the problem?

bitbank2, 2025-04-03:
Nevermind - I found and fixed the problem. I'll let you know when I push the new code.

Chris01011964, 2025-04-03:
Thanks...if you need a sample code let me know

Chris01011964, 2025-04-05:
Hello Larry 

the issu still exits but became a little bit better. With the Lib 2.8.1 you have been able to use a image in the command with a size of 22K. The new lib allows a max of 6K before when the error occurs.

I have added a simple program just paint a yellow box on the screen. you can play with the values "SpriteWidth" to generate the error. 

Kind regards
Christian 

E (6431) spi_master: check_trans_valid(1053): txdata transfer > host maximum

#include <bb_spi_lcd.h>
#include <esp_heap_caps.h>
#include <esp_psram.h>

BB_SPI_LCD tft;
BB_SPI_LCD sprite01;

#define WIDTH  480
#define HEIGHT 272
#define TFT DISPLAY_CYD_543

int SpriteWidth = 60;   	                               // Startwert
int SpriteHight = 100;                                    // Beispielhöhe

uint16_t *Im_sprite00 = (uint16_t *)heap_caps_malloc(SpriteWidth * SpriteHight * sizeof(uint16_t), MALLOC_CAP_SPIRAM);

void setup() {

  Serial.begin(115200);
  while (!Serial);

  tft.begin(TFT);                                        // Display initialisieren
  tft.setRotation(2);                                    // Orientierung auf Querformat
  tft.fillScreen(TFT_BLACK);                             // Bildschirm leeren

          
  sprite01.createVirtual(SpriteWidth,  SpriteHight, Im_sprite00);

}

void loop() {

        // Just paint a yellow box on the Screen

         sprite01.fillScreen(TFT_YELLOW);
        // 
        // Sprite 
        //
          Serial.println(SpriteWidth*SpriteHight);

          tft.pushImage(10, 10, SpriteWidth, SpriteHight, Im_sprite00, DRAW_TO_LCD);
          delay(100); 

}

bitbank2, 2025-04-05:
I believe what you're seeing is a memory leak, not a problem with the maximum transmission size. You're pushing pixels from SPIRAM in a perpetual loop. Try allocating the buffer in DMA memory and see if it fails.

Chris01011964, 2025-04-05:
ok....just wondering it works prfect with 2.8.1

Chris01011964, 2025-04-05:
There is no different if you use simpl internal RAM. Old lib 20K.......new version 6K.  Maybe I do something total wrong:-)  Anyway...I can work with the old lib:-)

bitbank2, 2025-04-05:
Are you using the code I just pushed? It sends large requests in 4k blocks to make sure it doesn't go beyond the 4k limit (of SPI/QSPI). Take a look. It shouldn't complain about large blocks.

Chris01011964, 2025-04-05:
No I did not...Where I can see it ?